atc1441/ATCwatch #1: Touch input is not working.

bc8483828, 2020-05-16:
Touch input works on the stock firmware and the P8TestMenu but not on P8TestMenuDisplay or the code here. 

I could help troubleshoot if you posted the P8Menu code here.

atc1441, 2020-05-16:
Hi. It is all the same code base, and i dont have the P8Menu code anymore. 

I changed the way it reads the Touch because of some errors with the interrupts. 
It sounds to me like the interrupts are not fired for the touch input. What device are you using? 
You can try to put the reading from the touch in the main display loop and see if it changes something. It will go a bit crazy as it will not delete the previous swipe gesture and will continiue to swipe.

bc8483828, 2020-05-16:
Ok, I will try that and let you know.  It seems like there is something
going on with the interupts because when I hit the side button it waits a
random amount of tine to vibrate. Sometimes right away and sometimes half a
second later.

 I have a LIGE brand watch.  It is listed as BW0128A-WCX-US on the box.

atc1441, 2020-05-16:
Ok so if you have a different watch then i can not support it per se by this firmware. The pinout will be different. 
Please first get the pinout via some test code to see what corosponse to what.

The Bootloader is compatible to many watches as display and flash is same but the Firmware not. 

But we can get there :)

atc1441, 2020-05-16:
Do you still know what firmware it showed?
MOY-***-1.*.*

bc8483828, 2020-05-16:
I think it is the same watch. It looks exactly like a P8 and the stock software looks the same. I did not get the code version before I flashed it.  I should have a second one comeing next week so I can look at that one.

Today I tried to change the display int in pinout.h to all the other open pins other than 28. No luck.

I also made an if statment in the main menu screen to display the date according to a dig read. It worked with the push button but not the display int (28.)

Any insite into what changed between the older code and current code would help when I get a chance to look at this again. 

Even without the touch input this is nice. I changed it to display the US format date and time and moved it all around on the main menu. I can now see the whole text message when it comes in on the main menu page.

bc8483828, 2020-05-20:
I got it to work by preventing the touch display from going to sleep in touch.cpp.  Not sure how this will change the batrery life but over the past 6 hours it does not seem to be a lot worse.

I also had to take the main screen down to only time because touch input was not reliable with eveything running. 

This watch probably has a slower version of the processor or something. 

I have confirmed that the pinout is the same with test code. 

Would be nice to have some more info on the touch chip but I can't find anything else other than the pinetime wiki page.

Maxvi33, 2020-05-27:
Hello bc848..
I have the same problem with you.. Do you have more information about this problem ? 
I'll try to debug but you are sure for slower version of the processor ? 

i purchase other watch to try to compare..  bad luck...

atc1441, 2020-05-27:
Oh the sleep can be the problem. 

Maybe disable it testwise. But the battery still needs to be drained one time after it.

atc1441, 2020-05-27:
I commented out the sleep in the code for now,

This will not drain more battery as the touchscreen goes sleeping in its own when not touched for 3 seconds.

bc8483828, 2020-05-28:
I disabled the sleep mode in touch.c and made it so that when I press the button when the main screen is up it moves to the other menu. You can swipe after that. 

Not home now but we could probably fork this code with my edits for watches that do not work with this code. I have also added some other functions. 

My current battery life is about 40-50 hours.

atc1441, 2020-05-28:
The main problem is that it isnt because of a hardware difference but a software problem . When reflashing via stlink the problem is gone

bc8483828, 2020-05-28:
If the touch goes to sleep anyway we could probably stay on this branch. 

Are you seing the up/down swipe not working even after the touch fixed?

atc1441, 2020-05-28:
I never had the bug myself so i only talk to other people having it and try to debug it

atc1441, 2020-05-28:
Maybe I had at one time but I was able to fix it by tapping hardly on the screen

bc8483828, 2020-05-28:
I see that if I tap harder on the main menu it works more often. I think
there is just to much going on on the main menu. The uP has to much to deal
with.

atc1441, 2020-05-28:
On my stock home menu it shouldnt be a problem, the cpu must be the same otherwise it also wouldnt work. There must be something else making problems :-/

atc1441, 2020-05-28:
Would love to have one that has this problem to search for the bug. 

Are all the ones with the bug closed ones?

bc8483828, 2020-05-28:
Yes my watch is closed but I got a new one of the same kind and a ST Link in the mail yesterday so I will be opening one today. 

Also I need to get the stock firmware version off the new one. Any idea how do I go about doing that?

atc1441, 2020-05-28:
To see the version, swipe down in main screen and press the i icon there will be the version MOY-****

When testing the new version please also first try ota update to see if it does work or not please

atc1441, 2020-05-28:
Just tested it on a Stock P8 and the Touch works as expected my version was

MOY-TEH5-1.7.7

bc8483828, 2020-05-28:
The watch is listed as a P8a. Software is MOV-TFK5-1.7.7

atc1441, 2020-05-28:
Ok so that means it is not the same watch

I will look into the TFK firmware and look for the pinout that should clear thinks up. Hopefully

atc1441, 2020-05-31:
The Repo is now updated to a version where the touch gets read differentely please try it out and give feedback if it works, so far multible people gave positive feedback

closed 2020-06-02 by atc1441