atc1441/ATCwatch #2: init_accl not feeding the watchdog causing bootloop

0x416c6578, 2020-06-02:
i have been trying to run some custom code and although a fresh clone compiles fine, a bootloop is caused by the init_accl function. i have nailed the problem down to the first do...while loop:
```c
do
  {
    delay(10);
    rslt = bma4_read_regs(0x5Eu, data, 0x40u, &dev); //Error happening here - the read is failing 
    v4 = ((unsigned int)data[59] >> 4) & 1;
    delay(1);
  }
while ( !v4 );
```
where the second line in the loop (rslt = ...) will have rslt as a non zero value (which means there is an error of some sort), causing this loop to go on forever and eventually triggering the watchdog to reboot the device. i must admit i don't know enough about the bma421 at the moment so i might be missing something obvious, and i think i may be since from the history of this repo, it looks like the owner's build (available in https://github.com/atc1441/DaFlasherFiles) doesn't have an issue since it boots fine.

atc1441, 2020-06-02:
Hi. This is correct and known, i dont feed the watchdog there as if so you would end in a bootloop. 

It is still on the todo list to restart the init process if something goes wrong line you encountert.

Also in the main repo a reboot on boot up happens from time to time.

The problem with the BMA421 is that there is no correct library available and everything needed to be reversed from the stock rom.

0x416c6578, 2020-06-02:
ok thanbks, just out of curiosity, how did you get your build to work? is there anything special you changed that fixes it, or does it change from device to device

atc1441, 2020-06-02:
It should just work if you compile the current repo, dont know if you know, i am the owner of both repos. The DaFlasher files P8TestMenu is fresh compiled from this Repo and works fine.

Did you tried it with the compiled stock ATCwatch firmware?

0x416c6578, 2020-06-02:
yeah the compiled stock firmware on DaFlasherFiles works fine, compiling this repo from a fresh clone doesn't.

atc1441, 2020-06-02:
Thats strange, i am running exactly that version on 3 devices right now. 

What device do you use? What arduino version/board?

0x416c6578, 2020-06-02:
i am using arduino 1.8.12, with the "DSD6 Tracker" board selected. i am flashing a new p8 watch from aliexpress (https://www.aliexpress.com/item/4000637492509.html). i am using nRF connect to flash it, but that shouldn't make a difference i dont think. i am compiling on linux, and i followed the instructions you gave for the heartrate lib (this works fine). i can comment out init_accl(); and it will boot fine, but of course the accelarometer or touchscreen dont work

atc1441, 2020-06-02:
Ok the reason seemse to be the board you selected, it needs to be dafit / p8 
It fails because the i2c is not working with the other board. 

Please use the Portable version of the arduino ide linked in this repo.

0x416c6578, 2020-06-03:
sorry to be a pain but can i just ask what files you changed to get i2c working in the portable version. i am trying to replicate the libraries in my own linux version, but i get the ble_gatts.h error caused by not having a softdevice. i have added the board to `boards.txt`, put the corret stuff in the variants directory, but i still get the error. i think i am missing something but ive looked and i cant find anything else that is different

atc1441, 2020-06-03:
It should be just the variant.h file from the P8 varaints folder that is making the pinout of the i2c and other stuff, but currently only i2c and Uart is used from there. 

Hope that helps. 
i dont use it on linux so i can not give that much help on it but it should work as sandeepmistry is able to do so

atc1441, 2020-06-03:
did you also installed the BLEpheriphials library?

0x416c6578, 2020-06-03:
i have figured it out. because linux is case sensitive, the S132 in platform.txt isn't recognized (it should be s132 like the D6 below it). also i had to change the adafruit-nrfutil line since it had .exe instead of just the binary (since it is in my path i removed the whole absolute path to it also). i am planning to write up what i did at some point because my whole toolchain is a bodge job that i cobbled together. cheers for the help though, and thanks for the work you have done with this

atc1441, 2020-06-03:
Nice to hear it :)

Sounds good if you can make a write up that helps the next person

closed 2020-06-03 by atc1441

0x416c6578, 2020-06-03:
i have made a writeup on my fork of this repo if anyone reading this is experiencing the same problems as i did