atc1441/ATCwatch #3: Compiling issues

sgungnir, 2020-06-19:
I had been compiling the pre-LVGL code but now it does not (post-lvgl).

I added the LVGL library to the arduino install but then I needed a config, I looked for it in your pinetime repo and added it to the project but still getting issues.

It looks like it needs some theme code ? 

Thanks !!!

sgungnir, 2020-06-21:
The lv_conf.h you are using is inside the library folder and without that compiling is not working. I tried modifying the one included in the library but it did not work.

Could you commit it along with the code ? THANKS !!!

sgungnir, 2020-06-21:
Figured out that you are using version 2.1.5, and I was not able to install that with the library manager, had to get it directly from the repo. Then copied some of the settings from your pinetime one and got it working.

It looks like everything is working but I am not 100% if I am missing any settings or changes, would love to have your exact config when you have a chance !

Thanks for all the work, it is great !!

atc1441, 2020-06-21:
Hi. 

Please use the D6Arduino.rar file linked in the repo.

There is everything what is needed

sgungnir, 2020-06-22:
I didn't know you updated the .rar, I had the version from before LVGL and didn't see a note about a new version.

It is working great.

Thanks !!

closed 2020-06-22 by sgungnir