atc1441/ATCwatch #4: Unable to flash (maybe broken on delivery) P8a

AlmightyFrog, 2020-06-27:
After long time of waiting and ordering two times, finally today my P8 arrived.
It was powered on and worked first few minutes, now it seems like touchscreen is not responding (still on stock).
Any idea, what makes that touchscreen issues? The button, step counter and also arm gesture to enable display works fine.

Sometimes it responds to touch inputs and then i can navigate and pair it with my phone, but that time was never long enough to use DaFlasher.

Is it "safe" to keep on trying flash the alternative bootloader? Information screen tells "MOY-TFK5-1.7.7", it this clock in meantime supported?

atc1441, 2020-06-27:
hi,

that could be a hardware or software problem.

It is known that sometimes the Touchscreen does not work unless it is taken from power, so either by discharhging or by disconnecting the battery.

You could try to flash the new firmware or wait one time till it is empty. 

It should be no problem to flash the other bootloader even without the touch working.

The Firmware is the Correct one and is supported

AlmightyFrog, 2020-06-27:
Not sure what was cause, but i got some steps further. After several reboots of smartphone and doing the manual disable/enable bluetooth like hinted it was recognizend and i got until flashing P8TestMenuDisplay.ino.zip succesfully.

So that is now on and also displaying some information, but no touchscreen interaction.
With some iterations of killing DaFlasher and disable/enable bluetooth it is recognized, but it does not start flashing ATCwatch.ino.zip I just built.

Flashing stops with line "Started DFU" and progress bar stays at 0%.

atc1441, 2020-06-28:
If the touch if not working even after flashing ATCwatch there is no other real chance of waiting till the battery is dead. 

To make it faster you can disable the check times in the main loop so the display will stay on.

Hope it will be working again after battery is drained. It could he of course also be a real Hardware problem, something like the flex cable is not correctly inserted or so

AlmightyFrog, 2020-07-01:
Strange behavior. I drained the battery, recharge a little and touchscreen worked.

Then I found out why newest version in meantime does not build anymore, after git bisect i found commit 0dd3138d10d5c8f1a776ad1b7f1d4819d686e46f which explains it. Maybe some change like that would be also interesting for Readme with a date and note like "You need to re download arduino from here".

Anyhow. Built new version, flashed it to clock and touchscreen stopped working. Not sure if cause of update or some other issue. I reflashed old version i had before but still no touchscreen anymore. So now I am back at current version waiting till battery is empty again.

Do you have any guess what could have happened?

atc1441, 2020-07-01:
I will add an FAQ into the Readme where i will add that you need to update the D6Arduino.rar sometimes, its possible that i forgot to add it there when a bigger update is there so its simpler to make i more generous, thank you for the tipp and i totally get the frustration when someone is not that involved in it.

That sound like it is the known problem with the touchscreen that it needs to be discharged to get it working again,

### To your problem:
But that it does not work in ATCwatch is strange, there where lately a few changes to the Touchscreen, see this video https://youtu.be/jgjKaSETY8Y 

The newest version sends an reset command every time the watch goes into standby and wakes it up on wakeup of the watch.

Please try to comment out the following two line to disable that sleep command:
https://github.com/atc1441/ATCwatch/blob/master/ATCwatch/sleep.cpp#L40
https://github.com/atc1441/ATCwatch/blob/master/ATCwatch/sleep.cpp#L68

this will disable the touchscreen sleep on standby.

Hope that helps

Also a nice info would be to get what touchscreen you got.

AlmightyFrog, 2020-07-01:
Already short note: I got the "bad" version as you claim it in your last video.
If battery is quite low, there is a higher chance for touchscreen to work some time.
Then it just stops working any you don't get it back mostly until battery is almost empty again.

I also saw that Touch Test going to 255x255 when pulling finger off and did want to make you a photo of it, but timeout for display is faster than i can take a picture. Anyhow you captured it already on video and mine just behaves like the black one on the right side.

I will do your suggested changes as soon as i get to be able to do them and then report here my findings.

AlmightyFrog, 2020-07-01:
Ok, i flashed the changed firmware some hours ago and tried from time to time reaction.
- most times touch did not work at all
- sometimes it did work, but then stop before i could navigate to touch menu
- sometimes it did work, even into touch menu, but then at some point the displayed events seemed strange and shortly after touch stopped working

I've got the feeling that either my touchscreen is broken or not reacting with events as expected by code.
So @atc1441, what would be the best/proper way to modify firmware to not show main screen but only always the touch debug screen.
Would it work to change in _display_home()_ variable _currentScreen_ to _touchScreen_ without side effects?

AlmightyFrog, 2020-07-02:
I've put clock on my desk and tried to not move it. Touch didn't work for first 3 clock "unlocks" (wait till display turns off and turn it on again), the next time it worked. Without issues for quite some time, even further unlocks.

Then out of sudden, in touch menu the behavior changed slightly before there were no more touch events shown:
Normaly it says something like Event=3,X=255,Y=255 if you do not touch it.
Slightly before no more events were shown there for some taps after lifting finger you could instead read Event=0,X=0,Y=0. If the taps itself did show the correct XY positions i can't tell, as that was too fast to read for me.

I already mentioned it indirectly before, but my Tp Version says on display to be "00 B4 00 02".

atc1441, 2020-07-02:
Hey, the effects you have sounds like the early stages where there where problems with the touch.

this should be fixed in the current firmware and i didn't got any other negative feedback so far and tested in for my self on different watches. 
But could be of course still be a firmware problem as it sounds so similar.

you can directly exchange this https://github.com/atc1441/ATCwatch/blob/master/ATCwatch/menu.cpp#L87 with the touchScreen without any problem.

After disabling the touchSleep cmd does the screen lights up after you touch it ?

AlmightyFrog, 2020-07-03:
I spent some time try to find out more, but am more or less no step further.

With unmodified SW: First it seemed touchscreen works directly after device was not used for longer times or battery is almost empty. But that could also have been only by accident.

With no touchSleep SW: If touchscreen works, you can wake up the clock with touching it, otherwise no reaction. Pressing hardware button still wakes up properly.

With touchScreen App only SW: If touch never worked so far, the Display is showing everything equals zero (Gesture. Touchpoints, Event, X, Y) besides Tp Version 00 B4 00 02 and Interrupt = 1.

atc1441, 2020-07-03:
Hard to help you further, 

do you have the chance to get a second P8 ?

AlmightyFrog, 2020-07-05:
In theory that was my plan, but it not only did take months to arrive also one P8 got lost.
One I did want to open to tinker around attaching debugger and the other one use for real.

So currently I do not plan to get a second one - e.g. the screen burn in effect is a little annoying and it seems like lottery whether you get the good or bad version.
Not sure if that is worth another 20€ and long time waiting to then again get the "bad" version.

Btw: Is there an IRC channel or similar existing for eased communication?

atc1441, 2020-07-05:
You can enter the discord server here: https://discord.com/invite/TRktBnv 

I understand the trouble with the bad version, maybe there will be an even better watch in the future so stay tuned.

AlmightyFrog, 2020-07-16:
Thanks for your support so far @atc1441, but currently I sadly don't find enough time to tinker around more. Maybe in a month or so i will join the discord channel.

As battery life is quite good, I might just flash the watch with default atcwatch firmware and live with only home screen. As it contains everything relevant so far (time, date, pulse, steps) i guess it's better than nothing.

That "better watch", what are you planning to maybe tinker around also in future?

closed 2020-07-16 by AlmightyFrog