atc1441/ATCwatch #8: Missing files

sgungnir, 2020-07-07:
font57.h is missing, also, the inc_tick function is not found. (I assume there is a missing file or it is in a changed file that was not commited).

atc1441, 2020-07-07:
It sounds like you dont have the newest D6Arduino.rar with the LVGL library, if you are on linux you can copy it from the D6Arduino.rar

The font57.h was needed before the LVGL and also inc_tick is inside of the LVGL

closed 2020-07-07 by atc1441

sgungnir, 2020-07-07:
Thanks for that ! Working late at night is fun, but sometimes dangerous :) What software are you using for source control ? do you just manually upload the changed files? or are you using the git client on your machine and making commits ? (just curious)