atc1441/ATCwatch #10: Add software license

AlwaysLivid, 2020-07-28:
Hello there, I was wondering whether you could license the source code of this repository in a way that allows people to make contributions to ATCwatch "legally". Thanks!

atc1441, 2020-07-28:
Hey, 
could you explain why it does not comply with Bosch's License ? 
It has the Copyright notice in every file from Bosch so i dont see a problem, but could be wron of course.

AlwaysLivid, 2020-07-28:
> Hey,
> could you explain why it does not comply with Bosch's License ?
> It has the Copyright notice in every file from Bosch so i dont see a problem, but could be wron of course.

Oh, my bad, you're right. I'm just used to seeing the LICENSE files on the root of a repository.

Still. If I were to fork this repository, you'd be supposedly able to take down my material. If you were to accept a pull request from another person with a tree that's not licensed, that could complicate issues further-- particularly because, for starters, you wouldn't be legally able to license your entire project, let alone how other factors at play, such as namely the distribution, would be legally ambiguous.

atc1441, 2020-07-28:
Thank you,

so i am willing to change the license of the project to this one: https://github.com/kemitchell/CC-BY-NC-SA-4.0/blob/master/CC-BY-NC-SA-4.0

this would prevent commercial use but lefts all other opens open i guess. what do you say about it?

AlwaysLivid, 2020-07-28:
> Thank you,
> 
> so i am willing to change the license of the project to this one: https://github.com/kemitchell/CC-BY-NC-SA-4.0/blob/master/CC-BY-NC-SA-4.0
> 
> this would prevent commercial use but lefts all other opens open i guess. what do you say about it?

An OSI-approved license would probably be more preferable from my point of view; https://opensource.org/licenses

I've seen code licensed under the Creative Commons licenses before, but I think they're more geared towards, well, "creative" content, such as different forms of multimedia or blogs. CC themselves recommend open software, but the first thing they show you on their website as a potential solution is the GNU GPL license. https://creativecommons.org/about/program-areas/software/

There was a discussion in the PineTime chat regarding licensing watch firmware the other day.

atc1441, 2020-07-28:
For future reference Added GPLv3 license to it.

closed 2020-07-28 by atc1441