atc1441/ATCwatch #18: SC7A20 accelerometer

goddade, 2020-08-17:
The model of my watch is p8b, stock firmware is MOY-TIN5.
The accelerometer of ATCwatch firmware does not work.
I found that the accelerometer chip of p8b is not bma421 but SC7A20.
Is there a SC7A20 version of ATCwatch firmware?

atc1441, 2020-08-17:
Hey, so far not, there is active Discussion going on in the Discord server https://discord.com/invite/TRktBnv

That Accl does not have integrated step counting so there is a real algo needed to use it thats a big step but needed as more and more Watches have the the SC7A20

goddade, 2020-08-18:
I want to write a driver for SC7A20, but I don't know the configurations of BMA421.
For example, data resolution, full scale, work mode, ODR.

atc1441, 2020-08-18:
Here are the configs i use. 
https://github.com/atc1441/ATCwatch/blob/0dd3138d10d5c8f1a776ad1b7f1d4819d686e46f/ATCwatch/accl.cpp#L36

The Interrupt function is not used of the BMA421 as it has no usefull features for the usecases.