atc1441/ATCwatch #26: D6 Notification APP Problems and desired additions

androidonis, 2021-04-30:
I see some issues with D6 notification APP. After installing and using the APP, I can not flash any new firmware to the P8. I am using nRFconnect by Nordic and it terminates firmware uppload with GATT Error at 5....40% of upload. (Unfortunately Da Flasher APP never starts a DFU-Upload and shows 0% for ever)

-The communication gets somehow disturbed by D6 notification.
-I try to stopp the D6 not. app, but it can not be stopped (Android-Settings-App-D6...-stop APP does not work - still running).
-I need to deinstall the APP, and after that nRFconnect can do its job.
-Re-Install ONLY works from Playstore. If I backup APK with APK Export APP the re-install fails (without WiFi), with WiFi it says ok, but on every 'connect' to P8, I get 'this app needs gets closed on errors'. I go to Playstore, install from there, and it works again.
- Please implement a STOP button for the APP.
- Another cool thing would be a BLE console, like the message log and on the same Tab a CUSTOM BLE CMD field. (Simpy put this field on the top page) 

It may, that some effects are related to my chinese tablet (Android 6.0 Kernel 3.18.19+), but in generall other Apps do not show similar problems.

atc1441, 2021-04-30:
Hey. You can disable the "D6 Notifications" that will disable the service and you will be able to flash the device

androidonis, 2021-04-30:
But even with notifications disabled, it seems not completely passive. Any idea what the problems are with 'not playstore' intallations from APK file?

atc1441, 2021-04-30:
What does "not playstore" installation mean?

androidonis, 2021-04-30:
First - thanks, it did run through with disabled notifications (at least just now).
I mean installation from APK file with android packet installer. Create a backup with the APK Export APP, De-Install APP, go to filemanager and select 'D6 Notification 71.apk" (my size 1.06MB), allow packet installer to install. This fails if WiFi is turned off and goes through with WiFi/Internet connected. But the installation is somewhat compromised and the program seems to crash on every connect. I just googled and found that this might be related to D6 notification being a multi APK bundle and you need a special installer for this type - true?

closed 2021-05-24 by androidonis