atc1441/ATCwatch #30: Script to build with arduino-cli on linux

elazarrosenthal, 2021-12-28:
Hi 

I spend most of my computer time in linux. 
I have been trying to get ATCWatch to build without needing to flip to windows 
I know you are busy on many projects so I went to work on it myself

I put together the following script  I put in file setup 
and call 
# bash setup testdir
Seems to work 
but could use some help from you 
the script needs to download all of D6Arduino j
just to get the lv_arduino library you used that is not available  anymore (v2.15)
also there is a properties json file you have that is bad (my conclusion I could be wrong)
I'd like to contribute this to your project if you can fix up and publish as part of it

Thank you 
Elazar Rosenthal elazar@panix.com elazarrosenthal@gmail.com

# give args should fail if $1 exists but we don't
if [ $# != 1 ]
then 
	echo "usage: setup newdir"
fi

#create build directory
Workdir=$1
mkdir $Workdir
cd $Workdir 

# setup needed librarys that we have to extract
mkdir -p Arduino/libraries/

#change HOME to be here to keep real HOME clean all files in this dir
export HOME=`pwd`

# install local standalone copy of nrfutil
pip3 install wheel --user
pip3 install adafruit-nrfutil --user
# add to PATH
export PATH=$PATH:$Workdir/.local/bin

####
#### get arduino-cli
####
#donwload
wget --no-check-certificate  https://downloads.arduino.cc/arduino-cli/arduino-cli_latest_Linux_64bit.tar.gz
# extract
tar xvf arduino-cli_latest_Linux_64bit.tar.gz  arduino-cli
rm  arduino-cli_latest_Linux_64bit.tar.gz 

####
#### get D6Arduino # we need for lv_arduino version that is not available anymore
####
# comment out for speed and use cp when testing 
wget --no-check-certificate  https://atcnetz.de/downloads/D6Arduino.rar
# copy local copy to speed up if we have 
# cp /home/elazar2/D6Arduino.rar .

# extract lv_arduino 2.15 (cant get from github)
unrar x D6Arduino.rar D6Arduino/portable/sketchbook/libraries/lv_arduino-master 
mv  D6Arduino/portable/sketchbook/libraries/lv_arduino-master  Arduino/libraries/.

# clean up  
rm -rf    D6Arduino/ D6Arduino.rar 

####
#### get HRS3300-Arduino-Library
####
git clone https://github.com/atc1441/HRS3300-Arduino-Library.git
mv HRS3300-Arduino-Library Arduino/libraries/

# get ATC WATCH archive
git clone https://github.com/atc1441/ATCwatch.git

./arduino-cli config init
./arduino-cli config add board_manager.additional_urls https://atc1441.github.io/D6Library/package_nRF5_boards_index.json
./arduino-cli core update-index

# this will fail due to error with file on web remove & and cp below wehn fixed
./arduino-cli core install nRF52D6Fitness:nRF5 & 

### due to error in index file on web the above will fail and we need to use a local copy
# remove wne fixed
wait
cp ../package_nRF5_boards_index.json .arduino15/package_nRF5_boards_index.json
./arduino-cli core install nRF52D6Fitness:nRF5 
#  remove above when fixed
./arduino-cli core update-index

###
### edit files to allow for .a from HRS code
###
# make backup of file to edit
cp .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt.orig
cp .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt.new

# add line 
echo "compiler.libraries.ldflags=" >> .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt.new
# append to line 
sed 's/combine.pattern.*/& \{compiler.libraries.ldflags\}/'  < .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt.new > .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt.new2

#  overwrite file with changes
cp .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt.new2 .arduino15/packages/nRF52D6Fitness/hardware/nRF5/0.7.5/platform.txt 

####
#### buildit 
####
./arduino-cli  board attach -b nRF52D6Fitness:nRF5:dsd6Watch  ATCwatch/ATCwatch/
./arduino-cli compile --clean -e -b nRF52D6Fitness:nRF5:dsd6Watch ATCwatch/ATCwatch/

`

`

elazarrosenthal, 2021-12-28:
sorry the above was supposed to be code quoted attaching file
[setup.sh.txt](https://github.com/atc1441/ATCwatch/files/7786416/setup.sh.txt)

elazarrosenthal, 2021-12-30:
I was mistaken about lv_arduino 2.15 its there (I forgot about tags I has been looking for a branch)

elazarrosenthal, 2022-01-03:
Ok 
I made my own fork and branch 
attaching new setup script with output also 
[setup.txt](https://github.com/atc1441/ATCwatch/files/7802877/setup.txt)
[setup5.out.txt](https://github.com/atc1441/ATCwatch/files/7802879/setup5.out.txt)

https://github.com/elazarrosenthal/lv_arduino/tree/atcwatch

full checkout build on my machine
[setup5.out.txt](https://github.com/atc1441/ATCwatch/files/7802872/setup5.out.txt)
[setup.txt](https://github.com/atc1441/ATCwatch/files/7802873/setup.txt)
 / internet connection takes about a minute
elazar@elazar-HP-ENVY:/work0/elazar/watch$ time ./setup watchtest5 > setup5.out 2>&1

real    0m58.146s
user    1m5.387s
sys     0m6.676s

atc1441, 2022-01-03:
Hey. 
Thanks a lot for doing it. 
I am currently not able to look into it to as the time is rare. 
Will do it in the future :)

elazarrosenthal, 2022-01-03:
I cleaned up the script and put here
https://github.com/elazarrosenthal/AtcWatchBuilder

elazarrosenthal, 2022-01-03:
Thanks for responding 
I know your time is valuable 
You do amazing stuff

elazarrosenthal, 2022-01-11:
Apparently someone else found my repo for the script 
and it worked for them  (mostly)
https://github.com/elazarrosenthal/AtcWatchBuilder/issues/1

stefan123t, 2022-02-02:
Dear @elazarrosenthal,

thanks for the script you provided. I am still new to getting this running on my P8b but so far succeeded to compile ATCwatch.ino under Linux by comparing the D6Arduino.rar with the various libraries available on the Library Manager. But the script you provide is much easier!

Here is a quick write up of what little I had to do using your script:
```
git clone https://github.com/elazarrosenthal/AtcWatchBuilder
cd AtcWatchBuilder/
./setup p8watch
```
If you want to edit the code afterwards you can do this in the Arduino IDE too:
```
cd p8watch/ATCwatch/ATCwatch/
arduino ATCwatch.ino
```

Kind regards,
Stefan

elazarrosenthal, 2022-02-02:
Thanks for the input

stefan123t, 2022-02-08:
@drtonyr can you provide some more detail on how you ran DaFlasher under Linux. I am currently stuck at this point in the recipe. I wonder if it would be possible to make the same changes that Aaron applied to the Nordic Semi Demo DFU Android Application to the nrfutil python code too. Eventually it is only some checksum verification that has to be ignored and simply send the DFU file after the init packet.

drtonyr, 2022-02-09:
@stefan123t 'can you provide some more detail on how you ran DaFlasher under Linux'?

I didn't run DaFlasher under Linux.  The work being discussed in this thread is 'Script to build with arduino-cli on linux'. you've still got to get the result to your watch which I did by emailing it to myself, saving the attachement to the phone and running DaFlasher from there.
'

stefan123t, 2022-02-09:
Okay, thanks that clarifies the details you were hinting at. Still have to get this step to run under Linux too, as I have no Android around.

x29a, 2022-02-09:
@stefan123t not to hijack this thread, but if you want flashing capabilities under linux check out the work [here](https://github.com/x29a/nrf52-OTA-BLE-Linux) and even better [fanoushs desay_dfu python](https://github.com/fanoush/ds-d6/tree/master/fwdump#docker) which even runs in a docker container.